ali1234/raspi-teletext #1: Suggestion

skinkie, 2015-02-08:
I'll test your software later on my Raspberry. Since my communication with broadcom I was under the impression this wasn't possible due to the fact that the vbi slicer was missing. We did something similar years ago for the nVidia cards. You mind find it interesting: http://ttxinsert.sourceforge.net/

ali1234, 2015-02-08:
Hi. I've seen ttxinsert before, it's been mentioned to me a few times. I was able to get 5 usable lines of text using a similar method, but having the VGA port generate PAL RGB + CSync directly. See http://al.robotfuzz.com/generating-teletext-in-software/

skinkie, 2015-02-08:
Thanks. Although TTX is kinda old, it is still requested a lot by people. I see if I can help you a bit at least with testing :)

closed 2015-02-08 by skinkie