ali1234/raspi-teletext #3: Not working on Pi B+ w/ Alba LCD TV

OfficialMoonlit, 2015-02-11:
The TV detects that a teletext service is available and I get the "P100  Raspberry Pi Wed 11 Feb " in the header line, but no clock or demo text. Same result with earlier build, before the demo text was added. Using a Raspberry Pi B+ running fresh stock Raspbian and Alba LCD TV, Pi output set to 4:3 and tired updating to Feb 07 firmware. Not sure where to start looking for clues, is my TV just not playing ball? Any suggestions on tweaks I could try?

Also tried with KWorld capture card, K!TV/DScaler and the stock KWorld tuner software "TiVme" report no valid pages, but that may be a result of the card and not the Pi/raspi-teletext.

ali1234, 2015-02-11:
Hi,

Almost certainly there is still something wrong with the demo because it doesn't work with alevt either. Carousel mode works fine though.

ali1234, 2015-02-11:
Should be fixed in latest commit.

OfficialMoonlit, 2015-02-11:
Sir, you rock, works a treat, the TV displays the demo just fine now. No such luck on the capture card, but I can't verify that it ever successfully received teletext via composite, so I'll chalk that one up to the card not being particularly good, I was only using it to test the TV results against anyway.

Thanks!

closed 2015-02-11 by OfficialMoonlit