ali1234/raspi-teletext #6: filler packets

ZXGuesser, 2015-02-22:
Since the filler packets are sent whenever there is nothing to send when the program is started reading from stdin (i.e. not demo mode), would it be better to send something other than the page 200 page header as this will interfere with trying to output magazine 2 packets.
As far as I can tell CEEFAX and Teletext both used packets with magazine set to 8 and row set to 25 with the row data all spaces for their filler packets since they only used magazine 8 for page subtitle rows on page 888 and broadcast data packets so that seems like standard practice, at least in the UK systems.

boomlinde, 2015-02-24:
That seems sane. I don't have dumps of local teletext, but I think this would at least be a sane default. Perhaps the magazine and page number is best left configurable?

ali1234, 2019-08-25:
Just merged a pull request which changes the filler to 8/25.

closed 2019-08-25 by ali1234