ali1234/raspi-teletext #7: Feature request: Wide Screen Signalling

jpmeijers, 2016-09-22:
I'm not even sure if this is technically possible, but it would be awesome if this software also did WSS.

See:
https://en.wikipedia.org/wiki/Widescreen_signaling

Use case:
I use a Raspberry Pi to create an analog TV channel via an AV to RF modulator. The teletext feature is already super useful. Having the ability to tell a TV to play the channel in 16x9 mode, using WSS will be the cherry on the cake.

ali1234, 2016-09-23:
This would be easy to do. You will lose even more lines from the bottom of the picture though.

ali1234, 2016-09-23:
Actually that's not true. WSS is under the teletext so you won't need to modify tvctl at all.

ali1234, 2019-09-06:
Closed by mistake :(

ali1234, 2020-02-18:
This should now be possible using config.txt `sdtv_aspect`. It didn't work previously due to a bug. See https://github.com/raspberrypi/documentation/blob/master/configuration/config-txt/video.md

commit 18fe7b80 (2019-09-06): Implement power level option

Closes #7