ali1234/raspi-teletext #8: question : regulating the scroll speed of sub-pages

shungerz, 2016-10-05:
Pardon my ignorance, but how do you regulate (i.e. control the speed) of sub-pages appearing ? I would like to control the appearance of sub-pages from 7 seconds to 20 seconds instead.

peterkvt80, 2016-10-05:
raspi-teletext does not control anything to do with subpages. It is the text stream that you feed raspi-teletext that governs this. VBIT happens to be fixed at 7 seconds which is what I assume you are using as a text stream. VBIT doesn't support variable carousel timing at the moment but I'll have a look at it this week.

shungerz, 2016-10-06:
Thanks, good to know. I'm actually using pyteletext as the streaming solution into raspi-teletext.