ali1234/raspi-teletext #9: Raspberry Pi 2

boomlinde, 2016-10-08:
The RPi 2 uses a different SoC (BCM2836) for which the peripheral base starts at 0x3f000000 instead of 0x20000000. I tried changing the `map_base` in `tvctl.c` accordingly, and the program worked fine with Raspberry Pi 2. I might submit a patch if I learn a clean way to detect the model of the unit, but for now I am just leaving this here for reference if anyone wants to run tvctl on Raspberry Pi 2

peterkvt80, 2016-10-08:
Haven't needed to do that since the code was updated a year ago. It works fine on all versions of Pi.

davidjnz, 2016-10-09:
Thanks for this boomlinde.  Any chance you could let me know the line to change in tvctl.c?

ali1234, 2016-10-09:
Line 142.

davidjnz, 2016-10-09:
Any chance you could paste what the line should be changed to?  Would this account for lines randomly disappearing and some corruption on the page?

ali1234, 2016-10-09:
map_base = mmap(0, MAP_SIZE, PROT_READ | PROT_WRITE, MAP_SHARED, fd, 0x3f807000);

ali1234, 2019-01-30:
Closing this since the current version should support all models of Raspberry Pi.

closed 2019-01-30 by ali1234