ali1234/raspi-teletext #10: Option to skip or select VBI lines for teletext insertion

premultiply, 2016-10-10:
For some applications it would be usefull to have an option to skip (blank - no teletext output) in VBI for teletext insertion. By default all VBI lines from 7 to 22 are filled with teletext at the moment.
For example skipping line 16 - 18 would allow to insert VPS (german data line application for VCR control and station id) and VITS test signals.
Or reducing the total count of teletext lines would limit the datarate for DVB/DTV encoding and tunneling of teletext.

ali1234, 2016-10-10:
This would be a useful feature for coarse speed control too, since fewer lines means subpages stay on screen longer.

There is one small problem though: the software cannot know which field is being displayed because the API does not support it. So to work around this it draws the same data into both fields and updates it every frame. This means you would need to skip lines in both fields.

There is a way to work around this called double weave but it is more complex to implement so I did not bother as it was unnecessary. But given this and other feature requests it might be time to do it.

Tachiorz, 2016-10-12:
https://github.com/Tachiorz/raspi-teletext/commit/9b014884116d81b4989f43c5fd2bc90435864937
It isn't pretty, I was lazy to make proper options, but it works

ZXGuesser, 2017-12-03:
I have a feature request/suggestion which is similar to this so am bumping this issue rather than opening another. I hope that's ok.

My suggestion (if it's technically feasible) is to create a new mode (enabled by a command line switch to maintain backwards compatibility) which takes a additional data before the 42 bytes of teletext in the input stream. This would indicate the television line on which to the packet should be transmitted, so that the software creating the teletext data has control over which lines are used rather than both that software and raspi-teletext having to both be configured with the same number of lines to maintain synchronisation of field counters etc.
I'm not sure how you would buffer the data to wait for the next field before overwriting the last. Perhaps the interface could specify that the originating software sets an unused bit as a flag to indicate the start of a new field (or frame if using the true line number and synchronising the interlaced fields were implemented).
The unused lines would need to be cleared completely and the clock run in sequence appended to the teletext lines as they're drawn so that the quiet lines remain quiet.

This could also be the basis of support for (almost) full-field teletext as it would mean that the software generating the teletext data is easily able to ensure that the broadcast service data packet remains in the VBI without having to count packets to remain synchronised (and again not have to configure both programs independently to both use the same number of lines).

I appreciate this is a somewhat overcomplicated approach to configuring the number of teletext lines. I have a tendency to want to come up with a solution that satisfies every possible future use case rather than just the problem at hand!

ZXGuesser, 2018-08-13:
I forgot that someone had already had a go at this, and did a version where both fields use the same lines. https://github.com/ZXGuesser/raspi-teletext/commit/39a328e81fafa40e626c9f375950854623209133
In mine I kept the command line interface backwards compatible. https://github.com/ZXGuesser/raspi-teletext/commit/3c402fef018c7736173d61dde387022016769fdf

ali1234, 2019-08-25:
A version of this got merged, thanks.

closed 2019-08-25 by ali1234