ali1234/raspi-teletext #12: demo page numbers

davidjnz, 2016-10-31:
Is it possible to specify the demo page number.  Currently the demo page is 100 and 101 but i would like to set it to be 500 and 501?

Tachiorz, 2016-11-17:
You have to broadcast one of the packets: 8/30/0, 8/30/1, 8/30/2 or 8/30/3 to specify initial page. And change `demo_buffer.h`

ali1234, 2016-11-19:
I wanted to avoid putting any encoding logic into this software, so you will need to manually change the hardcoded arrays and recompile. Encoding and decoding belongs in a separate library which I am still working on.

ali1234, 2019-09-04:
Going to close this as the demo is just that - not meant for production use, just to ensure it is working. You should pipe in your own packets from an external program if you need something different.

closed 2019-09-04 by ali1234