ali1234/raspi-teletext #13: signal level

Tachiorz, 2016-11-16:
If I'm not mistaken, high level is max white now? Shouldn't it be lower?
ETS 300706 "1 Level: 66 ± 6 % of the difference between Black Level and Peak White Level."
Like :
```
#define RGB565(r,g,b) (((r)>>3)<<11 | ((g)>>2)<<5 | (b)>>3)
static unsigned short palette[256] = { 0x0, RGB565(164,164,164), 0xf000 };
```

ali1234, 2016-11-19:
Yes, probably, although it is very unlikely that any decoder would actually care.

peterkvt80, 2016-11-28:
The reason for using a lower level of white is to reduce overshoot, clipping, ringing and general distortion in the broadcast chain. The AGC will allow the decoder to cope, but it would take longer to adjust. This does cause loss of packets in some situations. None of this is important when transmitting the length of a SCART cable.

ali1234, 2019-08-25:
This could be made in to a command line parameter if anyone wants to take a look.

commit a3c19c00 (2019-09-06): Implement power level option

Closes #13
closed 2019-09-06 by ali1234