ali1234/raspi-teletext #15: Full Field / Channel transmission

m0yellow, 2017-03-12:
I love your software,
and I saw others requested VPS/WSS I would chirp in and like to ask if you could support full channel teletext transmission. 

In the ETSI Standard it is under .4.2 and Annex F.6. A TV Set then becomes automatically a teletext Monitor, you do not need to switch on teletext, and you get much faster transmission.

As with the other fields requested, it is useful to switch this property at runtime independently from the content transmitted. This would allow the encoder to transmit content for non teletext capable receivers. Or switch transmission on all lines on and off independently from the "auto-start" flag.

peterkvt80, 2017-03-12:
The trouble with this would be finding a full field decoder. It was very popular in the 1980s but most of this went away when the internet came along. Farnborough Airfield shut down their service about two years ago. I don't know what happened to their full field decoders.
If you look at typical decoder chips they are not capable of capturing more than 16/17 vbi lines. However, if you know of something that can do full field then it would not be very difficult to transmit full field.

m0yellow, 2017-03-17:
thanks for your insight. I saw a teletext only channel on a panasonic or sony CRT TV about 10 or 15 years ago. I was very fascinated that such a standard mainstream tv did decode the channel.

It was in use as info service on a local TV cable network in the austrian alps. There might be still some of this systems around, for tourist info and hotel networks.

From the speed and that I couldn't disable teletext I was pretty sure it was a full field teletext channel. The tv behaved normal (and slow) on any other channel.

I still have access to such an old TV set. we could try it, and setup my raspberry pi for a date with the old TV.

ali1234, 2019-09-04:
Full field would be useful for training with Domesday Duplicator as it would mean we can fit about 20 times as much training data on a tape.

commit 12e36bf1 (2019-09-06): Implement full-field mode

Closes #15
closed 2019-09-06 by ali1234