ali1234/raspi-teletext #17: Some TV models are frequently dropping packets or displaying corrupt packets

stojy, 2018-01-15:
Using your awesome project I've created a test page that alternates the contents of every character once a second.  In turn the test page is connected to 3 TVs simultaneously (via an AV modulator and 3 way splitter) so that each TV should show the identical content.

And most of the time it works flawlessly!  One of the TVs always works perfectly (Sony), but 2 of them (both TCLs) will 'randomly' every 3-5 seconds either..
- Drop packets, i.e. visible because entire screen changes except for 1 row (sometimes 2 rows).
- Show corrupt packets, i.e. visible because some characters have various bits flipped.

The two TCL TVs have no problems displaying 'regular over the air' teletext.

Looking through the code and the distro, I've tried tweaking various parameters (alpha blending, screen offsets, reduced packets per frame from 12 to 5, palette colour, etc) but nothing seems to help.

Is this a known issue?  Any suggestions on how to track down what's going astray?

a-v-s, 2018-01-27:
Could this be related to the signal level? #13

peterkvt80, 2018-01-27:
You might be right. The text signal is set to peak white and black. When you drive a square wave signal to the limit you will get overshoots, ringing and possible aliasing. The standard uses a smaller dynamic range and a raised cosine filter to avoid this. I suspect if you alter the palette it will start to work much better.

stojy, 2018-01-29:
After a lot of soul searching, I eventually found the issue.. it was HW, not SW.  One of the TV power supplies!

One of the 3 TVs had a dodgy power supply which was causing some 'back channel' signal distortion that not only affected the TV with the dodgy power supply, but also affected the other 2 TVs which were effectively 'connected' via the 3 way analog splitter.  FYI, the distortion was visible once I displayed a white background on the screen as you could then see the 'wavy lines' on screen.

Since swapping out the power supply, all 3 TVs now work perfectly without a single fault :)

Regarding the palette, I did review #13, but found that anything less than ~85% RGB565 started to make things worse.  I know it sounds counter intuitive, given the spec requirement of 66% and square wave with cosine filter.  When I get a spare moment, I'll setup a 'USB CRO' that I've borrowed to take a closer look at what is going on.

Thanks for the help.

closed 2018-01-29 by stojy