ali1234/raspi-teletext #18: question - why is width set to 370 pixels?

stojy, 2018-01-15:
I can piece together most of the code (albeit I've guessed my way through some of the dispmanx calls), but I'm stumped as to why the width is set to 370px.

45 characters * 8 + 8 offset = 368px.  But if I use 368 as the width, the TVs no longer understand the teletext signal.  It's presumably used to create the 144ns /pixel timing requirements (ETS spec), but I can't work out the math :(

I'll sleep much better if someone could please tell me the answer?

Also, I'm suspecting it may possibly causing some scaling 'artefacts' when the image is later stretched to 720px that could account for #17.

ali1234, 2018-01-27:
The value was determined experimentally. It isn't quite exactly what you would expect (54us / 375 = 144ns) but as I recall that value did not work. Probably the Pi's PAL timings are a little bit off and I had to compensate for that.

I didn't have a scope when I wrote this. I will re-check it to see if it can be improved.

stojy, 2018-01-29:
That makes sense.  After scratching around in the math, we (my colleague @pcalin) figured it must of been derived experimentally.  We had tried with a test page (but no scope) other values +/- 1-5 and found 370 was the sweet spot.

However you might find this interesting, we found that we got better results (i.e. less corruption.. refer #17), if the image was scaled down instead of up.  i.e. create a 370*2 image and scale it down to 720, instead of scaling up 370 to 720.  Presumably dispmanx produces less scaling issues going down.

closed 2018-01-29 by stojy