ali1234/raspi-teletext #20: GPU lockup

ZXGuesser, 2018-10-12:
I've been having a peculiar problem with the system I'm building for the imminent Block Party event next weekend. The system has four pi zeros each outputting video with teletext. Every so often the video of a pi will freeze up and GPU stuff hangs until the teletext process is killed (at which point video playback etc resumes).
Once the teletext process has been killed the following line appears in the kernel log
`vchiq: 0: ignoring ERROR from callback to service`

I've been running my own fork for individual assignment of VBI lines through most of my testing trying to narrow the issue, but I've checked the upstream code this evening and the same freezes occur.
I'm playing out video using omxplayer, but the same freezes happen whether video is running or not (and don't occur during playback without teletext active)

ali1234, 2018-10-13:
I think I know what the problem is.

Try commenting out this line: https://github.com/ali1234/raspi-teletext/blob/master/teletext.c#L150

See the bug report in the comment above for why it is there - but the bug should be fixed now.

ali1234, 2018-10-13:
Alternatively if that does not work then likely the bug is not actually fixed in the firmware and there isn't much I can do about that. The vsync callback was always really buggy if two programs were using it at the same time and it is very likely omxplayer uses it and will freeze up if it does not get a vsync callback.

ali1234, 2018-10-13:
It is also possible that this is a regression so try older firmware and kernel.

PR #24 (2019-09-04): Vsync rewrite
This refactors the dispmanx code into a single source file which can be shared between cea608 and teletext binaries. It changes the vsync handling to be more reliable and fixes some other small problems.

This should fix #20 and #22 and make the code work properly on RPi 4.

I need someone to test the cea608 code please.
closed 2019-09-04 by ali1234