ali1234/raspi-teletext #22: No output with cea608 on latest RPi  on 3+

unwiredben, 2019-08-06:
I updated my Raspberry Pi 3+ to latest raspian stretch today, and now I don't get any output from the cea608 program.  I spent some time debugging, and I'm not seeing any calls to the vsync callback.  Any ideas?

ZXGuesser, 2019-08-06:
Yes, the current Raspberry Pi firmware has broken it. I rolled back to a version from March (with the hash 00bb2894c0a4349464c19ba62f62b8d9e17a1a9e) using rpi-update.

unwiredben, 2019-08-06:
Thanks, @ZXGuesser, I can confirm that downgrading to that firmware restores correct operation.  I also just tried the latest firmware via rpi-update and that also seems to work.

ZXGuesser, 2019-08-06:
That's good to know!

ali1234, 2019-08-20:
I have a potential fix for this here: https://github.com/ali1234/raspi-teletext/commit/b8695eca93226f3c149cbf499262f0131f79f722

It does fix the deadlock I experienced on recent firmwares. Can you try it and see if it has any side effects please?

ali1234, 2019-08-20:
Note that I didn't patch cea608 but you should be able to just run teletext without tvctl and see on screen if it is updating. Let it run for a while... it takes several seconds for the deadlock to show up.

ali1234, 2019-08-25:
I have rewritten the dispmanx code to be shared between the two programs: https://github.com/ali1234/raspi-teletext/tree/vsync-rewrite

Please test cea608 now.

ali1234, 2019-09-04:
Hopefully fixed.

closed 2019-09-04 by ali1234