ali1234/raspi-teletext #30: Piping data from python

git-vcs, 2021-05-31:
I am trying to pipe-data into the program using python with the – option. I tried using one of the predefined pages from the demo, but it does not look like it’s working. The code runs without problems but the tv is not decoding what I am sending inn.
I tested the demo-program and it works great!
I tried piping “line bye line” and a hole page at once.
I have not used python to control other processes before, so I am not sure where the problem is.
You also mentioned something about the “clock run-in”, is that something I need to provide?

```python
import subprocess
import os
import time
import numpy

## data from demo
testPage= numpy.array(..........)

def teletext():
    program ="./teletext -"
    s = subprocess.Popen(program, stdin=subprocess.PIPE, shell=True)
    time.sleep(5)

    while True:
        print("writing")
        s.stdin.write(testPage.tobytes())
        time.sleep(1)

if __name__=="__main__":
    teletext()
    

```

peterkvt80, 2021-06-01:
You can't just send a page. You need to send a continuous T42 stream. That is a raw teletext stream without the clock run in and framing code. See how the demo.c pages are generated and work out how to do it in Python.

ali1234, 2021-06-01:
vhs-teletext contains routines for generating pages in python. See example here: https://github.com/ali1234/vhs-teletext/blob/master/examples/service

git-vcs, 2021-06-05:
Thank you, it turns out that I was not far off, I forgot to specify the right type of binary unit in the array.

```python

import subprocess
import os
import time
import numpy

testPage= numpy.array([],dtype=numpy.uint8)
  
  
def teletext():
    program ="./teletext -"
    s = subprocess.Popen(program, stdin=subprocess.PIPE, shell=True)

    while True:
        for line in range(len(testPage)):
            s.stdin.write(testPage[line].tobytes())
        

if __name__=="__main__":
    teletext()
```

closed 2021-06-05 by git-vcs