ali1234/raspi-teletext #31: No cea608 output on RPi 1

Famicoman, 2022-02-02:
Hello,

I've been trying to run this today and have two Raspberry Pi model 1s: an original with 256M of RAM and a Rev2 with 512M of RAM.

I can verify that both have working composite video output via RCA jack. I have verified that both have 64M of video RAM allocated and both have the latest firmware via `rpi-update`, though I also tested before updating as well with what was likely the original firmware on both devices.

On both, I have logged in as root and cloned the repo, then run `make` followed by `./tvctl on` followed by `./cea608`. It builds successfully on each `make`, `./tvctl on` always reports that cea608 is now on, and `./cea608` always runs and stays running until I kill it with ctrl-c.

I have tested with two different televisions, both on CC1 and don't see anything come up other than the console I am already on.

Happy to provide any more information or try anything further.

Thanks!

ali1234, 2022-02-03:
Unfortunately I don't have any way of testing the cea608 support. Nor any specs about what data should be sent. You'll need to send correctly formatted data before the TV can do anything with it. If it isn't correct, it will just be ignored.

Famicoman, 2022-02-04:
Thanks for the reply. Based on that comment am I to assume that the entirety of cea608 support in this application is experimental and untested? The Readme made it seem like the demo mode should work in displaying something.

ali1234, 2022-02-05:
Pretty much, yes. Some people claim to have used it? I don't know if they succeeded...

grahamparks, 2022-02-24:
I had a similar problem with UK Teletext. No signs of life on two TVs, an oscilloscope or HDMI. I found this note on the VBit2 instructions which solved the problem. Possibly it needs mentioning on this project too?

> If you are installing on Raspberry Pi OS "Bullseye" (released 2021-10-30) you must modify the boot config file to instruct the OS to use the old proprietary graphics driver from previous versions. This is achieved by commenting out the line dtoverlay=vc4-kms-v3d.

ali1234, 2022-11-15:
Is this a pull request? :)

ali1234, 2022-11-15:
I am not actually sure if this will work because dispmanx delivers vsync interrupts at 50fps ie per field. But there is no way to know which field it is. So there's a 50% chance it will go in the right one each time you restart the program. It can also flip if vsync deadline gets missed for some reason. This could be fixed by double-weaving the packets, but I don't think I implemented that.

ali1234, 2022-11-15:
Hmm... it looks like I implemented double weave for teletext but not cea608. On the latter, each buffer contains two packets and is shown for two fields. That means they might be delivered out of order, but it probably doesn't matter under that system as the two fields contain independent data.

ZXGuesser, 2022-11-15:
> Is this a pull request? :)

It can be 😄

ZXGuesser, 2022-11-15:
> I am not actually sure if this will work because dispmanx delivers vsync interrupts at 50fps ie per field. But there is no way to know which field it is. So there's a 50% chance it will go in the [...]

Yeah I found later that the issue on the firmware repo just got ignored re. vsync deadlines.
It seems to frequently lock up on restarting it with the failed update asset and needing a reboot like teletext used to.

ali1234, 2022-11-15:
Not sure about that. It used to work... even on pi 4.

ali1234, 2022-11-15:
Thinking of rewriting this in C++ so that there can be less duplicated code...

ali1234, 2022-11-15:
Please do a PR for the cea608 fix and I'll merge it asap. It should be fine wrt the field nonsense.

ali1234, 2022-11-15:
Fixed in https://github.com/ali1234/raspi-teletext/pull/32

closed 2022-11-15 by ali1234