ali1234/raspi-teletext #33: cea608 issues

rscullin, 2022-11-16:
I haven't been able to get cea608 captions to display with the recent update, even when testing with multiple different displays. While there's a non-zero chance it's operator error, I found what might be an issue in the caption data generation.

With a "standard" boot config, I get this output:
 
[image]

The caption data line is resized by dispmanx horizontally using a Hamming upscaling method, as mentioned in https://github.com/raspberrypi/firmware/issues/104#issuecomment-9585095 , which results in a blurry data line.

Adding the line `scaling_kernel=8` to `boot.config` to use Nearest Neighbor scaling results in an output that looks much closer to other examples I've seen:

[image]

However, even with that change, I'm unable to get a TV and a projector to decode captions. It might help with some other borderline decoders that are more forgiving.

I've tested on a "Raspberry Pi 2 Model B Rev 1.1" , running an up to date Raspberry Pi OS Lite, firmware version `921f5efeaed8a27980e5a6cfa2d2dee43410d60d`, `Linux raspberrypi 5.15.78-v7+ #1599` .

I used [`raspi2png`](https://github.com/AndrewFromMelbourne/raspi2png) to capture Frame Buffer data.

`boot.config`:
```
sdtv_mode=0
scaling_kernel=8
#dtoverlay=vc4-kms-v3d
#disable_overscan=1
```

Let me know if there's anything else you'd like me to try / debug.

ali1234, 2022-11-17:
How did you run it? Did you run tvctl first? What is your data source for captions?

The filter kernel won't be the problem. CEA608 is designed to work with VHS tapes.

IoIxD, 2022-11-19:
I came across the issue myself and noticed that it seems to work intermittedly.

tvctl needs to be on first, yes, perhaps an error should be thrown when its off. It also wouldn't work when PAL was off, turning that on seemed to fix things. I also made sure to disable dtoverlay.

And after 30 minutes of fiddling with settings to try and get it working, it worked!

And then I stopped it and tried to run it with an output and got this:
```
gavin@raspberrypi ~/raspi-teletext> cat /dev/random | ./cea608
cea608: render.c:36: render_thread_func: Assertion `update' failed.
fish: Process 6259, “./cea608” “cat /dev/random | ./cea608” terminated by signal SIGABRT (Abort)
gavin@raspberrypi ~/raspi-teletext> 
```

And launching it regularly doesn't work either until I restarted. and when i restarted, it again didn't work the first time, and redo it.

Passing stuff through stdin generally doesn't seem to work.

rscullin, 2022-11-21:
> How did you run it? Did you run tvctl first? What is your data source for captions?

Yes, I ran:

```
$ sudo ./tvctl on
CEA608 output is now on.
```

And then ran `./cea608`, expecting to see whatever the `cea608` demo is appear on the TV. I've tried re-running it dozens of times in case there's a timing issue, with no luck.

> The filter kernel won't be the problem. CEA608 is designed to work with VHS tapes.
Ah, good to know!