ali1234/raspi-teletext #34: Omxplayer and raspi-teletext

peppi001, 2023-04-23:
Hi,
It's only a question, not a real issue :)

Are there any solution to play a video and raspi-teletext at the same time?
Omxplayer is working, but the top of the picture is not visible and bottom has a black border.
Is it possible to play video with teletext without moving up the played video?

THX

ZXGuesser, 2023-04-23:
Moving the display upwards by 32 lines is how raspi-teletext is able to draw teletext signals in the blanking interval so the TV can see it. This means you lose the bottom 32 lines of the picture and there's no easy solution to that unfortunately.

Using `--win 0,32,720,608` in your omxplayer command line will move the back to the correct position.

I use omxplayer with the following arguments on my teletext pis `-b --layer 2 --win 0,32,720,608 --aspect-mode letterbox -o alsa:hw:0,0`

peterkvt80, 2023-04-23:
One solution is to get a real inserter. You can plug in a vbit-pi-3 on top of the Pi so the inserting is done in hardware rather than on the Pi. https://www.nathanmediaservices.co.uk/vbit-pi/

ali1234, 2023-04-23:
As far as we can tell the GPU does support this. It is not possible with the proprietary driver though. Only with the new DRM driver and requires extensive refactoring.

peppi001, 2023-04-23:
Thanks for the answers. 
ZXGuesser omxplayer arguments is good.

nutki, 2023-06-13:
@peppi001 
I have a similar use case where I want to use omxplayer with raspi-teletext. I solved this issue by allocating fewer than 32 lines to the teletext data. This requires two changes:
* allow shifting display by a smaller amount. I wrote a replacement for `tv_ctl` which supports this (https://github.com/nutki/rpi-music-television-simulator/blob/main/player/sdtv_ctl.c - with `-pos n` parameter).
* allow arbitrary `height` in `teletext.c` (I added code to control it via an additional `-h` parameter).

On the TV I am using for testing 4 lines of each field are not visible anyway so I can use 25% of the bandwidth without sacrificing screen estate (still need to offset the omxplayer display for proper alignment). 25% is more than enough for a couple of pages I want to generate.