ali1234/raspi-teletext #35: Raspberry Pi 4 support

halamix2, 2023-06-12:
tvctl can't be enabled on the latest RPi4

```sh
sudo ./tvctl on
Error mapping register memory.
```

Linux 6.1.21-v8+ aarch64

sdtv_mode=2
vc-kms-v3d overlay disabled
enable_tvout=1

halamix2, 2023-06-12:
changing the first mmap to not fail fixes the issue

cwgh1, 2025-04-22:
I can confirm this is still a problem on Raspberry Pi Model 4 and legacy OS "Raspbian GNU/Linux 11 (bullseye)" (using either 32 bit or 64 bit, lite or full).

There is no problem if I change the hardware to Raspberry Pi Model 3B.

halamix2, 2025-04-23:
I was able to get it running on RPi4, I don't remember if I still have the updated doce anywhere. I remember the issue is that the code fails if it cannot map register instead of trying the next one

cwgh1, 2025-04-23:
I switched to a Pi 3 Model B+ where it worked fine, so am using my Pi 4 Model 4 for something else now (an excuse for more Pis!).