ali1234/raspi-teletext #37: `tvctl on` distorts image in NTSC

nutki, 2023-06-15:
The change in 8d7bd58 causes `tvctl` to shift the even field display area up by 1 instead of the previous 2. Since the odd field is still shifted by 2, this causes a misalignment in the video display.
I have a [PR](https://github.com/ali1234/raspi-teletext/pull/36) that attempts to use the proper lines without modifying `tvctl`.
Maybe @ZXGuesser could verify this works as I don't have any device to decode CEA608 CC.

ZXGuesser, 2023-06-15:
Yup that seems to be working, at least as far as the demo code repeating "hello!" is concerned with the datastream going in the first field.

csboling, 2025-09-27:
The patch is working for me to display captions on my LCD TV, and doesn't affect the display as the current main branch does. Interestingly I am only able to get captions on the digital display, my two CRT displays do not display captions when set to CC1 (with or without the patch).