ali1234/raspi-teletext #38: [Enhancement]  Produce PES for injection into DVB TS

SonnyWalkman, 2023-10-27:
Hello @ali1234,
Is there any chance to allow the output to be DVB compliant and produce 42 byte PES to inject into a DVB TS? 

I'm using a Python app to convert 2 static T42 pages 898 and 899 and inject using TSduck into DVB-T TS which and works well.  The secret source in with ttx_gen.py produces a timed PCR to sync for TS muxing . I'm developing in go with is similar to C and could modify the code to suite. 
I'm looking to be usable on both Linux and Windows.

The idea of a few dynamic pages similar to the raspi-teletex demo is appealing.  Updating the header with machine date and time and produce T42 output via std or UDP pipe.

Here's is what I'd envisage for dynamic pages.

Page 100 Teletext Index, 
Page 102 Weather which gathers weather from external weather site Underground etc using API key. 
Page 104  Stats  bars graph for CPU, GPU and RAM usage temp etc. 
Page 105 (Fish tank)  https://www.youtube.com/watch?v=zFkBHuchOnk

[ttxgen.zip](https://github.com/ali1234/raspi-teletext/files/13185125/ttxgen.zip)

ali1234, 2023-10-28:
This is completely out of scope for raspi-teletext. It could be added to vhs-teletext as an output format but that's python anyway.

Why not just adapt your python script to read from stdin and pipe through it?

ali1234, 2023-10-28:
If you are just looking for some dynamic pages, vhs-teletext has a whole API for building them and a couple of examples.